jotego/jtcores #905: twin16: [darkadv] visible sprite line

jotego, 2025-01-01:
As the monster falls down, a red line becomes visible and falls down together.

[image]

The monster is in sprite SDRAM. Rendering the scene with fake 32'hAAAAAAAA data from the sprite SDRAM should reproduce the problem ok.

jotego, 2025-01-22:
trying to save the object RAM contents in [this branch](https://github.com/jotego/jtcores/tree/issue905)

niknakniknak, 2026-09-29:
This is legit behaviour https://youtu.be/ssnN7ytzbec?si=RwXSy4SKYbvnOGO8&t=624

closed 2026-09-30 by jotego

skutis, 2026-10-01:
But I still think we should try get missing in action working on the twin16 platform. The issue in #929

niknakniknak, 2026-10-01:
@skutis I've looked at 929. Played the game through. Can't see any ghosting

skutis, 2026-10-01:
Perfect, thanks for looking into this!

skutis, 2026-10-02:
BTW, I can't see any MRA file for M.I.A. twin16 version. Did you really play that version or was it the Turtles/Aliens version?
M.I.A. - Missing in Action (version R) (Japan), is the version I'm looking for.

jotego, 2026-10-02:
Again, we didn't do any direct thing to fix the ghosting so if that's gone is a side effect of having fixed something else for a different issue

niknakniknak, 2026-10-02:
yes its the twin16 version. heres an MRA for you to test with.

[M.I.A. Missing in Action (Japan, version R) TWIN16.zip](https://github.com/user-attachments/files/32957616/M.I.A.Missing.in.Action.Japan.version.R.TWIN16.zip)

skutis, 2026-10-02:
Thanks, I will try it. Sorry for doubting you. You are thorough.

niknakniknak, 2026-10-02:
no worries, I try to be.