jotego/jtcores #907: Simson: PCM issue in Vendetta

retrorepair, 2025-01-03:
I'm not sure if this is present anywhere else (I'd assume so) but if you run the Vendetta intro, you can hear the orchestra hit sample as the cityscape is scrolling past, however on the next section which introduces the fighters, there is a tone played which I assume is a YM tone instead of what should be the orchestra hit.

I again assume that the PCM is played using the YM as the note carrier so then the PCM is just absent and it's falling back to the YM synth.

For clarity, on this section the main theme notes should be the orchestra hit.

I can provide examples if required, MAME doesn't exhibit this issue so can be used as reference.

I worked on this board a LOT in my days working for arcade operators, probably 20+ boards and heard that intro many many times.

jotego, 2025-01-19:
Reproduced in simulation with `jtsim -video 2000 -w 200`
Added channel-enable control to jt053260 in 44928324e for debugging.

jotego, 2025-01-20:
After carefully listening to it, I think it all comes down to the low-pass filter that is not present on MAME. The core implements the filter than can be seen on page 14 of the schematics available online. That filter has this frequency response:

[image]

I think that this filter is attenuating some frequencies to a point that might made them hard to hear for all but the youngest people.

I think the filter should be left in the core, as it was in the original.

rp-jt, 2025-09-08:
On Discord, the user @pepalacant has reported sound fx on Vendetta being distorted

@jtmiki can you check this in today's compile all? Thank you

jotego, 2025-09-20:
The sound distortion is noticeable in the demo (commit debadf1fe). These actions have no effect:

- reduce the volume in mem.yaml
- gate cen_fm with the PCM ok signals
- remove the RC filters

It can be heard in simulation with `jtsim -video 3500 -w 3300`

jotego, 2025-09-20:
Adding a limiter to the ADPCM mix solves the issue.

rp-jt, 2025-10-03:
User @PepAlacant on Discord reports that now the volume balance between FM and PCM is not always correct, so now the FM has lost some "sounds", for there are parts sounding too high or too low sometimes

For example, music test nº OE

retrorepair, 2025-10-05:
https://youtu.be/OQLKgyYrfH4?si=bArJ8g-af2RQLU4d

This is the sound that's missing FYI. I can check the latest version later to see if it's still the same.