jotego/jtcores #910: Crime Fighters - Low drum volume

Areku-The-OF, 2025-01-06:
One of the highlights of this game is its music and for some reason the drums are barely audible compared to what I remember. I've compared it with footage from MAME, Final Burn and even real hardware and the drums sound loud and clear on those.

jotego, 2025-01-07:
The mem.yaml description is referring to a passive adder, where rsum represents a resistor to ground:

``` YAML
audio:
  rsum: 10k
  channels:
    - {name: fm,  module: jt51,     rsum:  6.2k,  rc: [{ r:  859,   c: 33n  }, {r: 1rout, c: 2.2n }]}
    - {name: pcm, module: jt007232, rsum: 12.25k, rc: [{ r: 33k, c: 470p }], pre: 0.28 }
```
This produces the following gains in hex:

```
    .g0     ( 8'h80 ), // fm  = 1.0
    .g1     ( 8'h24 ), // pcm = 0.18
```

But in the circuit, Rsum (R9) is the feedback resistor of an opamp. So it has the opposite effect. 
[image]

A very recent addition to JTFRAME makes it possible to specify this behaviour by setting **rsum_feedback_res**:

``` YAML
  rsum: 10k
  rsum_feedback_res: true
  channels:
    - {name: fm,  module: jt51,     rsum:  3.2k,  rc: [{ r:  859,   c: 33n  }, {r: 1rout, c: 2.2n }]}
    - {name: pcm, module: jt007232, rsum: 13.25k, rc: [{ r: 33k, c: 470p }], pre: 0.28 }
```
Note that there are some changes to the channel rsum values too, but the critical part is the changed to feedback resistor. Now the channel gains are:

```
    .g0     ( 8'h37 ), // fm  = 0.43
    .g1     ( 8'h80 ), // pcm = 1.0
```

@rp-jt, please try the core with this change. I am not very confident about jt51's rsum=3.2k, because the two stereo outputs are connected. If the FM/PCM balance sounds off try making jt51's rsum 3.2k*2 and 3.2k/2.

Check out the schematics for the other games in the core and verify that they have the same resistor values for all components and that the analog circuit overall is identical.

rp-jt, 2025-01-08:
@jotego 

In 525c9e7f737772d439a741fc2b69194f04e07a3f, the pcm is quite loud, I think a bit too much
In bfa8adf977852e6cbe064e86af6efa8e56873b56, the pcm can barely be heard. For this commit, the gains are:
```
    .g0     ( 8'h80 ), // 1.00  fm
    .g1     ( 8'h11 ), // 0.13  pcm
```

Revising schematics for the different games, there are subtle differences between some resistances values:

In Aliens, referring to the previous screenshot, R3 and R7 (also coming from fm) are 5.1K, and R24, R25 are marked as 0

In Super Contra, Rsum=27K, equivalents to R3,R5 are 2.7K, R24,R24 do not exist. 

Thunder Cross is like Super Contra, except R3=10K and R5=22K

Also, the opamp intermediate between VREF and SREF does not exist in Super Contra nor Thunder Cross

jotego, 2025-01-08:
We will need to sort out how to set different gains for each game. Put that aside for now.

I think the problem with the PCM sound is in this line:

https://github.com/jotego/jtcores/blob/2145b5e7fca0fa7d719ce1fdd1fc072489c3a4eb/modules/jt007232/hdl/jt007232_channel.v#L69

That makes an unsigned to signed conversion but it adds an extra bit.The conversion goes as follow: 

- the unsigned input range 0~40~7F = 0~127 is transformed to
- the signed range C0~0~3F = -64 ~ +63

It should be ok to drop the MSB and keep using `[6:0]` instead of `[7:0]`. So one bit can be removed from all the 007232 modules. Re-run the unit test in jt007232/ver/gain too.

The line in audio_mod.yaml needs be modified to 11 bits too:

`jt007232: { data_width: 11, vpp: 1.0       }`

And the sound modules in each core will need a width alignment at the output port too. Cores affected: aliens, castle, flane, fround, mx5k, tmnt, twin16.

Check out that the sound does not get suddenly broken. The extra bit may have been needed to get the substraction done right.

jotego, 2025-01-12:
Fixed in b4c7c54

closed 2025-01-12 by jotego