jotego/jtcores #919: gaiden

Tano987, 2025-01-10:


jotego, 2025-01-11:
The system has an overscan that does not fit through the analog output on any of my screens either. I am not sure if your screen will show all of it either even if we allow further movement.
I limit the range to +/- 8 lines for several reasons. I'll keep an eye on this as I work more on the core but it may not get better for your screen.

closed 2025-01-11 by jotego

Tano987, 2025-01-11:
It's enough that you keep it in mind.
If it's possible to fix it, perfect.
If not, I'm still grateful for your intention.

Thanks.

mabrilG, 2025-03-21:
For the IremM92 core, wickerwaka developed a good solution for these games that have a lot of overscan, perhaps could be included.

https://github.com/MiSTer-devel/Arcade-IremM92_MiSTer/commit/7eea0a6e2cc1b1e3297b0ac42610138800bab1d8

It actually looks great.
Thank you for all your work @jotego

jotego, 2025-03-21:
Thank you. I see he's doing some sort of 16-line averaging to shrink the video. Maybe enabling the ASCAL scaler of MiSTer in mister.ini will have a similar effect.

I do not plan to shrink the analog video digitally as that opens up a box of worms when some lines became invisible as part of the information loss.

Thanks again for sharing it.