jotego/jtcores #926: Maybe Typo Issue that implied failed to call make_commit_macro during the build core process

wwark, 2025-01-13:
Hi,

https://github.com/jotego/jtcores/blob/1066afbcf1bd31fa4317c6bc873bffe416167bb6/modules/jtframe/src/jtframe/macros/macros.go#L33

Thank you for your time for the check

jotego, 2025-01-13:
The build process health is reported [fine](https://github.com/jotego/jtcores?tab=readme-ov-file#compilation---jtframe) in master. So I think your problem is local to your setup. If you give a bit more context, we may be able to help.

wwark, 2025-01-13:
I used debian 12 with golang with go version go version go1.23.0 linux/amd64

I have this issue

```
[/home/wark/Dev/fpga_dev/jtcores (master)]$ jtcore cps1 -mister
/home/wark/Dev/fpga_dev/jtcores/modules/jtframe/bin/jtcore: line 319: cd: cfgstr.hex: Not a directory
panic: Cannot retrieve git commit. Using 'unknown' instead.

exit status 128

goroutine 1 [running]:
github.com/jotego/jtframe/macros.make_commit_macro()
/home/wark/Dev/fpga_dev/jtcores/modules/jtframe/src/jtframe/macros/macros.go:278 +0x1b3
github.com/jotego/jtframe/macros.MakeMacros({0x7ffcc2331099, 0x6}, {0x7ffcc23310a9, 0x6})
/home/wark/Dev/fpga_dev/jtcores/modules/jtframe/src/jtframe/macros/macros.go:215 +0x24b
github.com/jotego/jtframe/cfgstr.Run({{0x7ffcc23310a9, 0x6}, {0xc000181040, 0x3d}, {0x7ffcc2331099, 0x6}, {0x7ffcc23310e8, 0x4}, {0x0, 0x0}, ...}, ...)
/home/wark/Dev/fpga_dev/jtcores/modules/jtframe/src/jtframe/cfgstr/cfgstr.go:40 +0x7c
github.com/jotego/jtframe/cmd.init.func4(0xc000182700?, {0xc00017f180, 0x1, 0x8})
/home/wark/Dev/fpga_dev/jtcores/modules/jtframe/src/jtframe/cmd/cfgstr.go:41 +0xf6
github.com/spf13/cobra.(*Command).execute(0xba85e0, {0xc00017f100, 0x8, 0x8})
/home/wark/go/pkg/mod/github.com/spf13/cobra@v1.5.0/command.go:876 +0x687
github.com/spf13/cobra.(*Command).ExecuteC(0xba8360)
/home/wark/go/pkg/mod/github.com/spf13/cobra@v1.5.0/command.go:990 +0x38d
github.com/spf13/cobra.(*Command).Execute(...)
/home/wark/go/pkg/mod/github.com/spf13/cobra@v1.5.0/command.go:918
github.com/jotego/jtframe/cmd.Execute()
/home/wark/Dev/fpga_dev/jtcores/modules/jtframe/src/jtframe/cmd/root.go:45 +0x1a
main.main()
/home/wark/Dev/fpga_dev/jtcores/modules/jtframe/src/jtframe/main.go:44 +0xd9
=============== JTCORE =======================
 mister compilation starts at 09:03:39 with seed 1 on wark-home
Explicit macros:
JTFRAME_RELEASE=1 
Log file: /home/wark/Dev/fpga_dev/jtcores/log/mister/.log
Quartus Prime Shell
Version 17.0.2 Build 602 07/19/2017 SJ Lite Edition
Copyright (C) 2017  Intel Corporation. All rights reserved.
Error: Expected arguments are --flow <flow_name> <project_name> [-c <revision>]
Error (23031): Evaluation of Tcl script /opt/intelFPGA_lite/quartus/common/tcl/internal/qsh_flow.tcl unsuccessful
Error: Quartus Prime Shell was unsuccessful. 2 errors, 0 warnings
    Error: Peak virtual memory: 756 megabytes
    Error: Processing ended: Fri Jan 10 09:03:40 2025
    Error: Elapsed time: 00:00:00
    Error: Total CPU time (on all processors): 00:00:00
ERROR: while compiling the project. Aborting
/home/wark/Dev/fpga_dev/jtcores/cores/cps1/mister
```

In make_commit_macro there is a reference to GetCommit() in common.go so maybe the include have some issue on my env.

The import is quite different of common in macros.go from other cases because of  the point (. "github.com/jotego/jtframe/common")

```
import (
	"bufio"
	"errors"
	"fmt"
	"log"
	"os"
	"path/filepath"
	"regexp"
	"strconv"
	"strings"
	"time"

    "github.com/jotego/jtframe/betas"
    . "github.com/jotego/jtframe/common"
)
```

jotego, 2025-01-13:
You have to run jtcore from a git repository. Do not copy folders around.

closed 2025-01-13 by jotego