jotego/jtcores #929: twin16: ghost sprites in miaj

jotego, 2025-01-14:
As the main character moves around, he leaves _ghosts_ behind. This can be seen in scene 1, where the character on the ground should not be visible. The one climbing up the stairs is the right one.

[image]

``` BASH
swcore twin16 ver/miaj
jtsim -s 1
```

skutis, 2025-01-14:
Maybe the sprite table is not emptied correctly?

skutis, 2025-01-14:
On real hardware it is emptied, it depends on how you have implemented the parsed sprite table scanning.

jotego, 2025-01-18:
A couple of notes from a conversation with skutis:

- objbufinit starts right at blanking. The 1-line delay was not correct
- the value used to clear the buffer is 8'h00, not MAME's 8'hff

jotego, 2025-01-23:
Skutis looked into this a bit more. If the CPU tries to access the object RAM while DMA is active, it will get locked because DTACKn does not get asserted for oram access if DMA is busy. 

@rp-jt modify twin16_main:

``` VERILOG
jtframe_68kdtack_cen #(.W(5),.RECOVERY(1)) u_dtack(
// ...
    .bus_cs     ( bus_cs    ),
    .bus_busy   ( bus_busy  ),
    .bus_legit  ( bus_legit )
//...
);
```

- include oram_cs as part of bus_cs
- when `oram_cs&dma_bsy`, bus_busy should also be asserted
- make `bus_legit=oram_cs`

This should fix the boot-up test

jotego, 2025-02-16:
I added the DTACK-dma_bsy locking as it was done in #946 to both twin16 and fround in df8c5e7c7. But I only implemented the write-lock on twin16 because it breaks the graphics on fround.