jotego/jtcores #931: Parodius Sound/Music Issue?

tunnotron3000, 2025-01-14:
I've been giving the Parodius core a go as it's one of my favourite games. When booting the core from the MiSTer menu, the game seems to function normally; however, if I reset within the core for any reason, when the game loads again, there's either no sound at all or the sound functions strangely. An example is a random piece of music playing in place of an item collection sound or the music fading out long before it should.

I've uploaded a video here:

https://www.youtube.com/watch?v=zOcfcWhDjs0

I'm using the MiSTer Pi Turbo Pack; everything is updated and I've even re-downloaded the game file via the script to eliminate the change that it was corrupted somehow.

Please let me know if you need any more information - thanks!

jotego, 2025-01-15:
confirmed on 9fae6e3

gyurco, 2025-01-21:
It's interesting, the main CPU generates an IRQ for the sound CPU, but how's the sound number is passed to it?

gyurco, 2025-01-21:
Looks like through the PCM chip.

jotego, 2025-01-22:
@jtmiki this is fixed.

closed 2025-01-22 by jotego