jotego/jtcores #938: Analogue Pocket Cores: 2nd controller is not recognized.

ruffianes, 2025-01-18:
None of the Cores recognise the 2nd controller. It is not possible to add credits or play. Before the latest updates it was possible.

jotego, 2025-01-18:
I do not think there has been any change in the framework code that should have this effect. We will double check it, but @ruffianes please double check on your end too. Maybe try a different controller.

ruffianes, 2025-01-18:
Hi. First of all thanks for the job that you're doing. 

I've tried with 2 different analogue pocket, 2 different docks and 8bitdo arcade stick, 8bitdo Neo geo and 8bitdo m30. All of them use to work well but after dec update the second controller it's not recognized by the cores. The issue is only with Jotego cores. Other cores like Neo geo, Megadrive, super nes, works without any problems.
I know there are more people with the same issue in Reddit. 

Thanks again for checking it.

jtmiki, 2025-01-20:
We have tested the following three cores in Analogue Pocket with two player versions. 
We used today's compile all.

jtriders...Sunset Riders
jts18......Alien Storm
jtsimson...Vendetta

Both of us could play all of these cores without any problems.
Commit: 4492832

rp-jt, 2025-01-20:
Currently, the way the cores work when you connect several controllers to the Analogue Pocket is through automatic detection and assignment. This is, when one of the action buttons is pressed, that controller is saved as Player 1, the following to press them is Player 2, etc.
For Player 1, the first controller detected by the Analogue Pocket goes through unless another controller is assigned and then takes Player 1's place instead. 
For the rest of the players, no input can go through unless a controller is assigned, this is, until the action buttons are pressed. 
This controller assignment is set since detection until reset.

@ruffianes does this behaviour still happen after you have pressed the action buttons on both your first and second controller?  Thank you for reporting

jotego, 2025-01-20:
@rp-jt I thought that behavior was only for the SNAC controllers. Is it for the regular controllers too?

rp-jt, 2025-01-20:
@jotego yes, it works that way for both regular and SNAC controllers for them to be able to be combined

Maybe, we could make it so, if no SNAC controller is selected the controllers are assigned normally ignoring the automatic detection?

jotego, 2025-01-20:
Let's wait for @ruffianes reply

ruffianes, 2025-01-24:
Sorry for the delay in my response. This has been a very busy week for me. Yes the issue still happens after press the action buttons.Basically, the analogue detect both controller and assign player 1 and 2 but cores only detect 1 player.

ruffianes, 2025-01-24:
The older versions of the cores didn't have this issue.

jotego, 2025-01-26:
Somes games only have one controller. Can you tells one specific game where it fails?

ruffianes, 2025-01-26:
Sure, pang, super pang, parodius, simpsons, alien vs predator are the ones I tried last week. 
Thanks

jtmiki, 2025-01-27:
We have tested the following games and both of us could play without any problems.
Pang, Super Pang, Alien vs Predator, Simpsons and Parodius.

However, it should be noted that Parodius is played by two players taking turns with one controller.

jtmiki, 2025-01-28:
We have just tested them again with 8bitdo controllers.
One is with wire and the another is wireless.
We could play well.
If you play Super Pang with 2 players, the second player has to choose the play mode.

closed 2025-01-28 by jotego

ruffianes, 2025-02-01:
Hey team. 
I've reset the 8bitdos and the pocket. Now after a while pushing 2nd controller button the core recognize it. The thing is the core swap controller 1 and 2. So definitely the behavior is weird. Hope you can fix it for the next update. All no JOTEGO cores works without any issues. 
Thanks again for your contribution to the community.