jotego/jtcores #939: Analogizer (Feature Request): add better SNAC game controller buttons mappin

RndMnkIII, 2025-01-18:
In some arcade cores as JTTWIN16 with Vulcan Venture for example the default SNAC game controller is not very optimal for playing the game. For example in Vulcan Venture that uses 3 buttons for the gameplay the default assignment of using A B X Y makes very cumbersome using main fire and secondary fire at the same time (pressing buttons B and X that are opposed vertically). 
What I propose is that the assignment be done more following a linear assignment of the buttons from left to right, then the button mapping should follow the order Y B A in the case of an SNES controller for games that use three buttons or the sequence Y B for those who use two, This would be more like the order of the buttons on a real arcade cabinet in which the buttons are usually in the same row.

[image]

jotego, 2025-01-18:
I agree that the default button assignment is bad for vulcan venture on that specific gamepad. But I do not think we can do something that will work well for Vulcan Venture and for all the other games supported. 

That is why most platforms have button remapping. I know you cannot remap SNAC buttons on the Analogue Pocket. I do not see a good solution for SNAC on Vulcan Venture.

RndMnkIII, 2025-01-18:
Assuming that arcade games employ a maximum of 6 action buttons, allow the configuration in crtcfg.bin to include a different mapping for the action buttons: 
Logical A -> Physical Y 
Logical B -> Physical A 
Logical C -> Physical B 
...

jotego, 2025-02-15:
Although having an alternative assignment may be good today, some game will show up again that is not good with that assignment. The only good solution is full remapping but that involves too much complexity for too little gain.
I'd rather not take any action on this one.

closed 2025-02-15 by jotego