jotego/jtcores #940: 1942/higumaru: Low frequency buzzing sound

joxheaf21, 2025-01-18:
After inserting coin a low frequency buzzing background sound appears.
If you do not insert any coins and just leave the game running, until it starts to demo the game play, the sound effects are clear and without the buzzing background sound.

jotego, 2025-01-19:
This is the frequency spectrum of the noise. It spreads out into clearly audible frequencies, so it may not be completely removable.

[image]

jotego, 2025-01-19:
This is a software bug. The CPU should have set all sound channels to period zero, but sets one of them to $FFF, causing the lowest possible sound the system can generate to be produced.

[image]

jotego, 2025-01-19:
A software patch to the ROM would be the best solution for this. I have no resources to execute this at the moment.

closed 2025-01-19 by jotego