jotego/jtcores #943: X-Men Konami Sound Balances/Mixing (with hardware recording)

danielkavka, 2025-01-19:
Hi Jotego.  I play X-Men on original 4p and 6p cabinets 2-3 times a month, so I have a good mental reference for what it sounds like.  (It's one of my all time favorite arcade games.) I have a couple of observations regarding the sound balance in the current MiSTer core.  
-
The first issue is the balance of the background music between the two sound chips. In MAME, the level of the FM section/synth/chip was originally too high relative to the PCM instruments compared to the original hardware, making the FM instruments too prominent in the mix.  This was originally required the user to manually adjust the FM playback level, but the more recent builds have corrected this to sound correct at default settings.  If you need a high quality reference, the Konami All-Stars album has X-Men recorded from original hardware with all the correct balances (and a bit of added reverb). 

I’m not sure what’s going on in the current beta core, as there are times when the FM instruments sound louder and other times when they sound quieter than they do on the original hardware. The guitar (heard in the Stage 1 Theme) does sound louder than it should be.  
-
The bigger issue (and one that’s also still a problem in MAME) is that the triggered PCM sound FX (impacts, explosions, mutant powers, anything that's not part of the BG music) are at a different level than the PCM music playback, and that balance can't be fixed manually by turning the FM instruments down.  
On the original cabinet, the music can still clearly be heard in balance with the environmental and impact sounds without being covered up. On most emulations. that I’ve heard, the sound FX are louder than the PCM section of the BGM, which results in the music (YM and K/FM and PCM chips) is regularly masked by the FX.  
Early example: in the core, the Stage 1 combat FX are louder than the PCM "Here Come the Heroes" speech at the beginning of the level, which is clearly audible on the original cabinet even with active combat noises being triggered at the same time.
If you turn the PCM channel down on MAME, the FX get quieter, but the PCM instruments and BGM/Voiceover samples are no longer in the proper balance with the YM synths and the music mix/voiceover level is no longer correct.  
From what I can put together, it seems like this is a problem with the K054539 emulation and how it handles the level playback of said "live"/triggered FX compared to the playback of PCM sounds in the BGM.
-
Lastly, the current core has some balance differences between the music and voices as well during cutscenes.  The main one I noticed is Professor X sounding much quieter than the music in the cut scene between Stage 2 and 3.  ("We have been taken by Magneto.")
-
There's a direct recording from original hardware on YouTube that preserves the original sound balances, and I can vouch that it matches my experience playing in person.  It's the best available reference for dialing in the sound, IMO.  https://youtu.be/qt0j7mPBUB0?si=8CrPo09YGVaYtVod

For tuning:
1.Go to 1:27.  Listen to the relative levels of the in-game effects (hits, mutant powers, etc) over the the "Here come the Heroes" speech.
2.Go to 3:27. Listen to the comparative levels of the in-game effects and "Pyro Will Turn You to Toast" vs the background music PCM track (orchestra hits, drums).
3.Go to 4:31. Listen to the comparative levels of the in-game effects vs the background music PCM track (record scratches, brass hits, drums, "X-Men" vocal Sample).
4.Go to 10:03. Listen to the level of Professor X’s voice compared to the background music during the cut scene.
5.Go to 10:19. Listen to the comparative levels of the on-screen effects vs the background music PCM track (bird screeches, stereo bird samples, drums, synth orchestra hits).

I know this is a lot of information, and I really appreciate your work.  As far as the gameplay and visuals go you've really created the best way to play X-Men outside of the original cabinets. :)

danielkavka, 2025-01-24:
I took another listen and made some more observations.  

The FM chip is definitely lower than it should be in the mix.  You can hear this on the Stage 1 theme where the melody in both the A and B section is played by the FM chip.  It’s way down in the mix to where you can barely hear it over the PCM instruments, where it would normally be clearly audible in the proper arcade mix

The PCM issue is more interesting because it seems like certain channels are out of balance with where they should be relative to the other PCM channels.  In MAME, for example, I now believe the problem is that the PCM channel that handles the sound effects is at a much higher level than it should be, and that’s why it drowns out the PCM music samples *and* the FM instruments in that situation.  

In MiSTer, the balance between the sound FX and the PCM music tracks seems to be better, but there are certain channels that are disproportionately loud/quiet/out of arcade balance in the mix.  In stage 1, the plucked guitar part is loud.  In stage 2, the electric guitar chords are louder than they should be.  During the attract intro, Magneto’s “X-MEN” is much quieter than it should be, similar to Professor X’s voice in the cut scene after Stage 2.

I haven’t done a critical listen farther into the game than that, but if I’m right there will be at least one or two PCM tracks per stage that are noticeably louder or quieter than their hardware board mixes.

danielkavka, 2025-03-11:
So bad news, there's some major music bugs starting on level 3 or 4 (I need to make a recording).  There's some sort of noise/percussion sample that gets triggered on a recurring basis, including in the credit roll where it's not tied to anything to do with gameplay.  I don't know if I just didn't notice it in the first two levels or if it really only kicks in later in the game.  I don't have an HDMI recorder but with a little rewiring I can make recording from the analog outs.

jotego, 2025-03-12:
Thanks for your reports. I still have not got to them, but I will.

danielkavka, 2025-03-14:
I need to double check this, but last time I also noticed that the stage music cuts out abruptly when you reach the level boss.  On the arcade board it fades out gradually.

real-amano, 2025-05-31:
Just for the record: Furrtek is currently reverse engineering the k054539 chip. Maybe some conclusions can be drawn from that work.

Garagefreek, 2025-07-20:
👍

jotego, 2025-07-30:
> Furrtek finished his efforts yesterday. The k054539 chip should be fully RE'd now.

Thanks for the heads up!

real-amano, 2025-08-02:
> > Furrtek finished his efforts yesterday. The k054539 chip should be fully RE'd now.
> 
> Thanks for the heads up!

Np. Since Run and gun AND X-Men are affected you probably know about the even newer 053252 work by Furrtek:  https://github.com/furrtek/SiliconRE/commits/master/Konami/053252

jotego, 2025-08-02:
Yes, Furrtek sped the K053252 work as I told him I needed it. He is the best!

jotego, 2025-08-09:
The K054539 is still not ready. What Furrtek had completed was the tracing part. He has identified the metal connections but still has not extracted the schematics for it. We'll have to wait a bit more.

Garagefreek, 2026-01-15:
> We are a small team and we are working on this issue actively. If we get more [patreon](www.patreon.com/jotego)/[github](https://github.com/sponsors/jotego) subscribers, we will be able to fund a larger team.

I understand and appreciate the time & effort being put in, you guys are doing a marvellous job!!

Regards.

closed 2026-01-16 by jotego

Garagefreek, 2026-01-16:
> This issue is fixed for today's release. The PCM chip implementation still lacks the DSP effects and that may cause a bit of unbalance byitself. There is a separate issue about DSP effects (#937) so the discussion about that will take place there.
> 
> The changes made to jt539 are consistent with Furrtek's RE work and apply correctly to the other two cores using jt539 (_Run and Gun_ and _Premier Soccer_)

Excellent work.