jotego/jtcores #945: Sunset riders High pitch note does not end after shooting gallery completion.

devilschile2, 2025-01-22:
Playing (the 2 players version EBD) When completing the shooting gallery on stage 3. There is a high pitch note that keeps playing during the score presentation. It gets muted in the beginning of the next stage.

Nothing extremely important but worth looking at when fixing issues.

jotego, 2025-01-23:
Are you referring to [this moment](https://youtu.be/0aL-U_fMPLo?si=Pp5YfITEhzILdoJm&t=420) of the game play?

devilschile2, 2025-01-23:
Yeah, that's the moment. i have to use wetransfer because  the file because it's too big.
[https://we.tl/t-LblK0Ul7KZ](url)

I noticed 2 things. I don't know if they are relevant but here they go. 

1. I played a couple of times when Sunset Riders was the first game being played after booting and it did not happen. Afterwards, I played Super Street Fighter II and then Sunset Riders. That's when I got the issue.
 
2. I would say the high pitched note starts playing in the beginning of the bonus stage, but that it's most noticeable in the end, because the rest of the music ends but the note doesn't.

Hope this is helpful.

jotego, 2025-04-22:
I have not been able to reproduce this on commit 8058f58bc. Thank you for reporting it but this might have been the side effect of something else not directly related to the core.

closed 2025-04-22 by jotego

devilschile2, 2025-04-22:
Just checked it out and you are right. I couldn't reproduce it myself, so it got fixed! Thank you very much for all the attention and effort all your team puts into all the fixes and awesome releases. Kind regards.