jotego/jtcores #958: Midnight Resistance corrupted graphics

RealCommander, 2025-02-04:
A few updates ago, I noticed midnight resistance graphics got corrupted.

jtmiki, 2025-02-05:
I tested it in MiSTer with a compile-all date of 24 January 2025 , Commit: fed8986, and it worked.
There was no graphics problem.
We intend to check it with compilall-all from other dates.

RealCommander, 2025-02-05:
This is happening on all my MiSTer setups

[image]

jotego, 2025-02-05:
The difference is in the colors. It should like this:
[image]

But version 73a563b looks like:
[image]

@jtmiki please find the commit that broke it.

jtmiki, 2025-02-06:
I have found the commit with a graphics problem.
Commit: c5c9e55 the date of 15/9/2024.

[image]

jotego, 2025-02-10:
The order of the GFX2 ROM files changed when moving to MAME0261 as the reference. It used to be:

``` XML
<!-- gfx2 - starts at 0xD0000 - length 0x80000 (19 bits) -->
<interleave output="16">
    <part name="fl09" crc="907d5910" map="01"/>
    <part name="fl07" crc="2068c45c" map="10"/>
</interleave>
<interleave output="16">
    <part name="fl08" crc="a936c03c" map="01"/>
    <part name="fl06" crc="b7241ab9" map="10"/>
</interleave>
```

But it became
```XML
<!-- gfx2 - starts at 0xD0000 - length 0x80000 (19 bits) -->
<interleave output="16">
    <part name="fl-09.18a" crc="907d5910" map="01"/>
    <part name="fl-08.16a" crc="a936c03c" map="10"/>
</interleave>
<interleave output="16">
    <part name="fl-07.15a" crc="2068c45c" map="01"/>
    <part name="fl-06.13a" crc="b7241ab9" map="10"/>
</interleave>
```

commit 06a58045 (2025-02-10): midres MRA gfx2 order restored. fixes #958
closed 2025-02-10 by jotego