jotego/jtcores #961: cps 1.5 tenchi wo kurau 2 Service Mode

gokeeper, 2025-02-06:
First, I’d like to thank Jotego for your incredible work on FPGA cores—I'm a new MiSTer owner, and I truly appreciate it.

I'm having trouble accessing Service Mode in Tenchi wo Kurau 2 (wofch), even though I set the option to ON. However, the world version, Warriors of Fate (wof), works flawlessly.

Is there any chance this could be fixed? Thanks so much!

jotego, 2025-02-06:
I think there is no service menu on this game. We need to remove the OSD option for this specific game on the JTCPS15 core.

jotego, 2025-02-14:
Modifying that OSD menu so it does not show like that in one of the games in CPS1.5 is quite a change and wouldn't be supported on all platforms. I'll leave it as it is.

closed 2025-02-14 by jotego