jotego/jtcores #962: Super Hang on on Analogue Pocket / MiSTer

Dawbo66, 2025-02-07:
Super Hang On MiSTer/Pocket seems to be missing the sounds when you over take other bikes, its silent, but using Mame there should be some sounds..

jtmiki, 2025-02-10:
I have tested it on today's compile all on MiSTer, commit: b506448.
As Dawbo66 said there is no sound effect when you over take other bikes.
On the other hand, there were sound effects when you were overtaken by other bikes.

Dawbo66, 2025-02-10:
If you crash you get the sound of them overtaking you, I'm not sure if your going slow and they over take though😁

jotego, 2025-02-10:
@jtmiki please check if this happens in old commits too.

jtmiki, 2025-02-14:
I have confirmed it in old commits.
Since the commit 324e62c on 30/8/24, this sound problem has occurred.

Dawbo66, 2025-02-14:
Is there a link I could download a working one until fixed please ?

Dawbo66, 2025-02-14:
I'm sure it has worked previously, can it be fixed ?

jotego, 2025-02-15:
The sound test screen operates correctly

[image]

Dawbo66, 2025-02-15:
Glad your looking into it, the game isn't the same without the proper sounds.

jotego, 2025-02-15:
PCM channel 8 is used for this sound effect.

closed 2025-02-15 by jotego

Dawbo66, 2025-02-22:
Thanks for the fix, i do find the volume of these sound effects quite low on both devices, especially the pocket, is there a way to adjust these ? thanks.

commit e2afe89e (2025-02-15): Fixes #962. May also be the fix for #243