jotego/jtcores #964: CPS1/CPS2 game input registering incorrectly

Rakurai-UK, 2025-02-08:
The core has been perfect since owning a Mister for the past 3 years. But only recently, there has been a problem in fighting games where special moves requiring motion inputs (quarter circle, dragon punch motion, half circle etc.) are not being interpreted correctly, resulting in inconsistent execution. It does not seem to affect charge characters as severely.

I performed the following steps to isolate the problems to the CPS1/CPS2 cores:

- Performed a clean installation on an empty SD card, downloaded the latest cores from update_all.sh
- Swapped out with another DE10 nano to rule out a possible defective board
- Plugged in controller directly into the DE10 nano's USB port
- Tested with 3 different USB controllers using different controller boards (PicoFightingBoard and Brook Universal Fighting board). The controllers are arcade sticks and a hitbox with no analog inputs with digital inputs only, operating in XINPUT mode. There are no functions that allow the directional inputs to run in analog mode.
- Played the console ports of the CPS1/CPS2 games in question, and encountered no issues - specifically SNES, Genesis (for the SF2/SSF2 series) PSX and Saturn (for Alpha, and VS. series). It is not a 'skill issue' - as I am already well established in the retro fighting games community.

jotego, 2025-02-08:
A couple of weeks ago, a change was introduced to use emulate analog control when using analog sticks. This should not have any effect if you do not use an analog stick, though.

Other than that, I cannot think of any change that should be relevant to this. You can access old versions of the cores [here](https://github.com/jotego/jtbin/commits/master/). If you can identify a version at which this starts occurring, and it consistently happens after that version, we can check what changed in the source code at that date.

It may also be related to MiSTer firmware changes as all inputs go through the firmware before reaching the core.

Rakurai-UK, 2025-02-08:
I've done another round of thorough testing.

This time I did a clean install of mr-fusion on a blank SD card. I copied the mame roms folder onto it, along with the _Arcade .mra files. This comes with the Mister software version 230423. (The one on my other memory card is on 241227).

I tested the latest and several old versions of the CPS2 core on both Mister software versions.

It occurs, that old and latest versions of the core running on 230423 works perfectly. When I play on 241227, the input problems come back. So it sounds like the latest Mister software broke something.

Rakurai-UK, 2025-02-08:
I tested with the version prior to 241227.

On version 240912 - the problem doesn't exist.

jotego, 2025-02-08:
Thank you for looking into it. You made a good finding. Could you move this issue to the [MiSTer repository](https://github.com/MiSTer-devel/Main_MiSTer/issues)? It is there where they can fix this. I do not work on the firmware myself.

If will help if you can reproduce the problem with a core in the [MiSTer organization](https://github.com/MiSTer-devel) and mention that core too.

closed 2025-02-08 by jotego

goronoguchi, 2025-02-19:
hi , 
I have an unusual  issue that occurs   with all the JT core  games ,  when playing 2P at the same time ,  games like 1943 , SSF2 or alien syndrome , when the second player stick moves left , the character moves left  as normal but at the same time  the first player  character moves right .So to sum up 
 when   player 2 stick moves LEFT 
player 2 character LEFT
 but ALSO player 1 Character RIGHT.
I m new to the mister community , and wouyl dlike to have help/ suggestion as how to solve that issue.
games based on other core like TETRIS are fine  , no control issue.
thanks a lot

goronoguchi, 2025-02-19:
I m playing with a MISTERCADE , using JAMMA supergun

jotego, 2025-02-21:
@goronoguchi, your setup seems to be sending multiple inputs, like both joystick and key signals. Note that JT cores always response to [keyboard inputs](https://github.com/jotego/jtcores#keyboard). Please ask about help with your setup in forums, discord, etc.