jotego/jtcores #968: [JTAliens] - No games aside from Aliens load on the pocket

TheJesusFish, 2025-02-10:
Tested all 5 games on a dandelion pocket.

- Aliens - Loads fine and plays fine
- Crime Fighters - Load bar to black screen
- Gang Busters - Load bar to black screen
- Super Contra - Load bar to black screen
- Thunder Cross - Load bar to black screen

jtmiki, 2025-02-10:
I have tested on today's compile all, commit: b506448.
I could play all 5 games in JTAliens on Analogue Pocket.

jotego, 2025-02-10:
@TheJesusFish try re-generating the .rom files.

closed 2025-02-10 by jotego

TheJesusFish, 2025-02-10:
@jotego No joy. Deleted everything in the JTAliens folder, ran Pupdate and chose to reinstall the aliens core. Everything reinstalled, but only aliens boots.

davewongillies, 2025-02-11:
@TheJesusFish try running pupdate/pocket sync again, the ROMs have been regenerated

TheJesusFish, 2025-02-11:
Ok, looks like the regenerated Roms have made it to the updaters. All 6 games work now.