jotego/jtcores #969: sf: sound effects have a chance to interrupt music playback in Street Fighter

Thelypody, 2025-02-10:
Street Fighter (US, set 1)
Running on MiSTer (Terasic DE10-Nano) via Analog IO Board.

On the current version of the jtsf core Street Fighter has an issue with audio playback. Throwing a punch or a kick has a chance of interrupting background music for a brief time, presumably caused by sound effects playing at the same time. This is observable both in normal gameplay and arcade's attract mode.

jtmiki, 2025-02-10:
I have tested it on today's compile all on MiSTer, commit: b656448.
As Thelypody said during a punch or a kick attack, the background music is interrupted.
This did not occur in MAME.
The referee's ROUND calls are also muffled and difficult to hear.

jtmiki, 2025-02-11:
I have found two commits of the same date that the core didn't work well.
I attach screen shots of these two commits. The date was on the 12/1/2025.
Commit: 94cbdd7

[image]

Commit: f3ba22e

[image]

jtmiki, 2025-02-13:
I have found the commit that broke the core.
Since the commit 31/12/24, 0b0559b, the background music has been interrupted during attacks.

jotego, 2025-02-14:
The sound gains changed in the file generated by `jtframe mem` at that commit:

The left one is the good one
```
    .g0     ( 8'h80 ) |   .g0     ( 8'h4F ), // fm
    .g1     ( 8'h1C ),|	  .g1     ( 8'h80 ), // pcm0
    .g2     ( 8'h1C ),|   .g2     ( 8'h80 ), // pcm1
```

commit f33dbffa (2025-02-14): sf: sound clipping fixed. Fixes #969
closed 2025-02-14 by jotego