jotego/jtcores #970: sf: alternative Street Fighter sets have non-working punch and kick buttons.

Thelypody, 2025-02-10:
Street Fighter (Japan) (protected)
Street Fighter (US, set 2) (protected)
Street Fighter (World) (protected)
Running on MiSTer (Terasic DE10-Nano) via Analog IO Board.

Alternative sets of the Street Fighter arcade all have the same issue related to button inputs. Core successfully maps three punch and three kick buttons, but they have no effect during gameplay. Directions, Start and Coin work perfectly fine, but without other buttons there's no attack options during gameplay.
Examining the service menu, shows that the game actually recognizes all six buttons (as shown on the screenshot below, Player 1 holds down all six buttons, causing six "1"s to appear in the first two columns). Interestingly, the test screen has a separate punch/kick display on the right which does not change at all for three alvernative .mras. Testing with "Street Fighter (US, set 1).mra" on the other hand shows that this display should change with button presses.

[image]

TheJesusFish, 2025-02-10:
Looks like this was fixed in the past? https://github.com/jotego/jtcores/pull/86

jotego, 2025-02-14:
Fixed in 44fa3a394

closed 2025-02-14 by jotego