jotego/jtcores #981: Bay Route: end sequence completely broken

oneoogeoo, 2025-02-21:
The end sequence (after the boss at the helicopter) was completely broken: only audio noise, black screen, followed by a red one.

TheJesusFish, 2025-02-21:
Confirmed. (Man that game is a bummer to play...)

jtmiki, 2025-02-25:
I have tried several times to reach the end sequence but I still can't reach there.
I will try it again.

oneoogeoo, 2025-02-25:
> I have tried several times to reach the end sequence but I still can't reach there. I will try it again.

you only need a lot of coins :)

TheJesusFish, 2025-02-25:
> I have tried several times to reach the end sequence but I still can't reach there. I will try it again.

Aside from bosses, I'm pretty sure you can just hold right and walk through the levels (assuming you keep feeding it quarters).

jtmiki, 2025-03-05:
Finally I reached the last stage.
After the protagonist defeats the last boss, the screen freezes at this attached screen where the protagonist approaches her. The music and audio are the same repeatedly. 
I tested it with yesterday's compile all. (Commit: 31abc75)

[image]

gyurco, 2025-04-18:
The problem is this instruction:
[image]

This one causes a state switch to E1 in the core, but not in MAME.

gyurco, 2025-04-18:
Restricting the state change to cmpi.l #data, Dx works.
According to mame:

_The state can then be modified by the
    program, executing the instruction
    CMPI.L  #$00xxFFFF, D0
    where xx is the state._

So restricting to Dx addressing mode should be OK.

The patch is simple:
```
diff --git a/cores/s16/hdl/jts16_fd1094_ctrl.v b/cores/s16/hdl/jts16_fd1094_ctrl.v
index aad98abe3..150de7379 100644
--- a/cores/s16/hdl/jts16_fd1094_ctrl.v
+++ b/cores/s16/hdl/jts16_fd1094_ctrl.v
@@ -71,8 +71,8 @@ always @(posedge clk, posedge rst) begin
             end
         end
         if( !op_n && !dtackn && sup_prog /*&& stchange==0*/ ) begin
-            // cmpi.l #data
-            if( dec[15:8]==8'h0c && dec[7:6]==2'b10 ) begin
+            // cmpi.l #data, Dx
+            if( dec[15:8]==8'h0c && dec[7:6]==2'b10 && dec[5:3] == 0 ) begin
                 stchange <= 2'b01;
             end
             // rte
```

Need to test some other games.

jotego, 2025-04-19:
That makes a lot of sense. Thanks!
@jtmiki please test other S16 games that use the FD1094. Also, mark this issue as completed.

closed 2025-04-19 by jotego

jtmiki, 2025-04-25:
I have tested following games on dfferent cores using the FD1094 and all of them worked well.

Outrun  
Turbo Outrun

S16     
Shinobi
Tetris
Passing Shot
Wonder Boy III

S16b    
Altered Beast
E-Swat
Flash Point
Golden Axe

S18     
Alien Storm
Bloxeed
Moon Walker

Commit: bdce866