jotego/jtcores #985: [Splatterhouse] Image shows a lot of distortion in HDMI

vanfanel, 2025-02-24:
Hi there,

With current core MiSTer version (just updated via update_all) HDMI image for Splatterhouse is full of distortion (colored horizontal lines) and comes on and off constantly.

Older versions worked fine.

Toryalai1, 2025-02-24:
I cannot recreate this issue. Tried with the latest mister main build and latest core

vanfanel, 2025-02-24:
> [@Toryalai1](https://github.com/Toryalai1) thanks for testing it. [@vanfanel](https://github.com/vanfanel) this could also be related to problems in your dc supply or thermal dissipation fan. [@jtmiki](https://github.com/jtmiki) could you test it too on one of the HDMI screens?

No. My setup is fine with regards to temperature. Way more demanding cores run here without problems.

Could It be related to dual RAM somehow?
My current setup is digital-only and dual RAM.

jotego, 2025-02-24:
It could be related to the dual RAM, but JT cores do not exercise it so I would expect it not to produce noise. The image suggests timing problems. As the core is passing the timing checks inside a given temperature and supply voltage window, I would expect the issue to come from a violation of that window. If you are confident about temperature, I would suggest using a different supply or removing possible switches or splitters that you may have on your supply cable. 

I have seen many instances of odd problems on the MiSTer that were caused by an supply voltage not stable enough or too low.

vanfanel, 2025-02-24:
> It could be related to the dual RAM, but JT cores do not exercise it so I would expect it not to produce noise. The image suggests timing problems. As the core is passing the timing checks inside a [...]

The problem only happens with `vsync_adjust=2`. With `vsync_adjust=1` there's no such problem.
All other JT framework based cores seem fine: only this one seems to have problem with `vsync_adjust=2`, and only starting to be like this recently.

For the setup stability theory: even the Saturn and Jaguar cores run perfectly here, and the Saturn one has done so for hours and hours.
As for the power supply, I use a Pi5 official power supply, which is know to be stable and good.
There's nothing strange with other cores either (this is not crappy 4C RAM, and the power supply is reliable too).

My theory is that maybe you did some change to the video timings recently? Can you test `vsync_adjust=2` on your side?

vanfanel, 2025-02-24:
> I'm running with vsync_adjust=2 and no issues. Single SDRAM and official de10nano

Then it could be my display. Huh, this wasn't happening before, but well, someone else will come reporting the same sooner or later since vsync_adjust=2 is a pretty common option.

Toryalai1, 2025-02-24:
Try to nuke your _Arcade folder, delete .cfg file related to splatterhouse and rerun update_all

closed 2025-02-27 by jotego

johhenrik, 2025-05-17:
"FIx" and explanation here: 
https://misterfpga.org/viewtopic.php?p=97479#p97479
https://misterfpga.org/viewtopic.php?p=98040#p98040