jotego/jtcores #987: s16b: Time Scanner (set 3, Japan, System 16B) (FD1089B 317-0024) does not boot

Thelypody, 2025-02-27:
Time Scanner (set 3, Japan, System 16B) (FD1089B 317-0024) from the _alternatives directory.
Running on MiSTer (Terasic DE10-Nano) via Analog IO Board.

Unlike other sets of Time Scanner, set 3 only displays black screen on boot. DIP settings appear to have no effect. This occurs on both analog and HDMI output.

jotego, 2025-02-27:
@jtmiki, please check this and find the latest working commit.

jotego, 2025-02-28:
I think you are likely to be missing some ROM files. Set your OSD warning time longer in the MiSTer.ini file so you can check the message.

Thelypody, 2025-02-28:
Unfortunately, I don't think that this is a rom issue on my end. I've temporarily replaced game roms with non-merged sets from MAME 0.275 romset and still get a black screen with no missing rom warnings on launch. For comparison, deleting roms outright and launching core without necessary files does trigger warning on launch.

[image]

This is my current rom setup for Time Scanner set 3, as far as I see all files mentioned in .mra are present and have appropriate checksums.

jotego, 2025-03-01:
I have fixed it. If you want to test it before the release, use this MRA file:

[Time Scanner (set 3, Japan, System 16B) (FD1089B 317-0024).mra.zip](https://github.com/user-attachments/files/19038652/Time.Scanner.set.3.Japan.System.16B.FD1089B.317-0024.mra.zip)

closed 2025-03-01 by jotego

Thelypody, 2025-03-01:
Everything is working as intended now, including the flip screen switch. Thanks!

commit cb27bf79 (2025-03-01): timescan3 boot up problem solved. Fixes #987