jotego/jtcores #989: jtframe crosshair support: clean up tasks

jotego, 2025-02-28:
- [ ] support for specifying an offset to X, Y coordinates using JTFRAME_LIGHTGUN_XOFFSET/YOFFSET
- [ ] crosshair overlayed video should have lhbl/lvbl delayed accordingly
- [ ] each crosshair should go invisible indepedently after 5 seconds of no mouse strobe inputs
- [ ] mouse-emulation via keyboard seems to be reversed: upside down/leftside right
- [ ] make crosshair coordinates visible through jtframe_sys_info, and update debug.md accordingly

jotego, 2025-03-01:
There was a file missing in d8ef6d8c after merging. I added it on 3f1bdcd81. But when I tested this on MiSTer, the video was completely broken. So I reverted the two commits. You can re-apply them on top of master and debug what got broken on MiSTer.

rp-jt, 2025-03-13:
#998 removed mouse emulation and added support for vertical games using the mouse

jotego, 2025-03-21:
solved in 3c2e9af58b2ae008645df0253e5990e96b57af60

closed 2025-03-21 by jotego