jotego/jtcores #994: ngpc: brown line appearing at the bottom of the screen

rp-jt, 2025-03-05:
Tested for commit 31abc75893f2108052617b8c119cc2f7d7731b3f on Analogue Pocket

tbalcom, 2025-05-26:
I am seeing the same in both ngp and ngpc cores. There is always a brown/orange horizontal line at the bottom of the screen. I tried changing the display modes to no effect.

[image]

jotego, 2025-05-26:
Isn't this issue duplicated?

tbalcom, 2025-06-01:
I'm still seeing the error after the 5/31 update.

jotego, 2025-06-01:
@rp-jt please revert that commit, it does not fix the issue.

jotego, 2025-07-17:
Let's fix jtframe_credits. I think HB/VB should be delayed too, not just the sync pulses. The digital video signals (HDMI/Analogue Pocket display) care more about the blanking than the sync pulse.

rp-jt, 2025-07-17:
>I think HB/VB should be delayed too, not just the sync pulses.

They are already delayed by `jtframe_credits`
https://github.com/jotego/jtcores/blob/09e08c610d01334021bda96d6365fc6d2f7b74a4/modules/jtframe/hdl/video/jtframe_credits.v#L416-L417

I'll add the delay for the syncs as well and test it on the Pocket

PR #1069 (2025-05-27): Fix for #994
Tested in a couple of games and seems to fix #994
commit dbf67c18 (2025-05-27): Fixes #994 (#1069)
PR #1111 (2025-07-18): Delayed HS/VS with jtframe_credits
Fixes #994 

Wait for new commit in Pocket before merging
closed 2025-07-18 by jotego