jotego/jtcores #1001: Super Hangon "MiSTer" using Analogue controls...

Dawbo66, 2025-03-14:
Shangon. MiSTer using Analogue controls when turning slightly left or right on a bend the bike flickers as if its trying to straighten up, so midway on the analogue stick causes this but a bit more left/right would be fully leaning over. It maybe as it should be, my only Ref is its not like it in mame, and using numerous analogue sticks it does seem to do it with all of them. Its best to try without turbo.. 

[image]

[image]

Dawbo66, 2025-03-14:
@jotego  
Edit... I have checked an older version and it works as i expected.
https://github.com/jotego/jtbin/blob/9504f53a1ca444e5b4b786a0c3fc73fd84a15824/mister/jtshanon.rbf

jotego, 2025-03-15:
I think the multiway joystick feature may be creating a problem here. We'll look at it.

Dawbo66, 2025-03-15:
> I think the multiway joystick feature may be creating a problem here. We'll look at it.

Thank you. 😁

closed 2025-03-21 by jotego

Dawbo66, 2025-03-21:
@jotego Thanks for the fix, its perfect now.