jotego/jtcores #1002: [JTALIENS] Misalinged sprites in the game "Aliens"

xDagmaRx, 2025-03-14:
After destroying the head of the 1st  boss the sprite is sometimes misaligned right down the middle.  Had a hard time reproducing this again.

At the end of the truck driving stage the door where the little girl gets captured has misaligned sprites.  Easily reproducible and seems to happen everytime. 

Tested using rom sets "Aliens (US) and "Aliens (World Set 1)"

[image]

rp-jt, 2025-03-25:
In simulation, it can be seen that the gate is made from two of the same object, where the right side of the gate is horizontally flipped. 

It seems like the first pixel in the flipped gate is not drawn

[image]

jotego, 2025-03-25:
Is it using the mirror feature?

rp-jt, 2025-03-25:
In jt051960.v  hflip and vflip are obtained directly from the object scan and are sent to jtframe_draw. 
It does not present a mirror property like it is the case in simson or riders for example

closed 2025-04-04 by jotego