jotego/jtcores #1004: Weird issue when building for MiSTer

turri21, 2025-03-17:
Ubuntu 22.04.4 LTS
GO 1.24.1
Every other requirement installed

On the exact same setup there were no issues building from the repo's clone about 2 months ago.
After cloning the latest repo (main branch) I'm getting this error:

jtcores (master)]$ jtcore gng -mister -d JTFRTAME_RELEASE --nodbg
Stopping at macro conversion

Any ideas?

terminator2k2, 2025-03-17:
shouldnt  JTFRTAME_RELEASE be JTFRAME_RELEASE? i dont compile JT's cores but just a heads up in case its a silly typo

turri21, 2025-03-17:
Oops nice catch but no unfortunately that was just a mispaste. The error exists on typing it right but thanks I have corrected my post above.

jotego, 2025-03-18:
You don't need to define the release macro explicitly. `--nodbg` already does that. 
That's probably not the issue anyway

jotego, 2025-03-18:
Make sure your are sourcing setprj.sh also try a fresh directory in case you have some files corrupted

closed 2025-03-18 by jotego