jotego/jtcores #1006: Glitch on right side of the screen on Pocket

rp-jt, 2025-03-18:
Reported on Discord. Can be seen in cores like paroda, simson and riders.

The error started in some commit of 28th February, I haven't found which one yet

[image]

jotego, 2025-03-19:
The image looks squeezed. The full content is there but it is as though the Pocket is trying to represent a wider image and it is filling the right border with garbage.

jotego, 2025-03-19:
The JSON files between JTBIN's commit on the 21st of Feb ([binaries here](https://github.com/jotego/jtbin/commit/7f1d60496bc7c3c0865506e3955815fed604266f) for jtcore's [commit](https://github.com/jotego/jtcores/commit/48161e9faf33dc1c1805de23b56eeeee5de14e6f)) and the latest one do not seem relevant. Check the rbf_r file of the 21/Feb using the latest JSON files to discard a problem in the JSON.

rp-jt, 2025-03-20:
> The image looks squeezed. The full content is there but it is as though the Pocket is trying to represent a wider image and it is filling the right border with garbage.

I think what can be seen is the same as in https://github.com/jotego/jtcores/issues/866 
In that case, it was due to a discrepancy between the video dimensions generated by the core and the ones stated in macros.def

> Check the rbf_r file of the 21/Feb using the latest JSON files to discard a problem in the JSON.

Ok!

rp-jt, 2025-03-20:
> The JSON files between JTBIN's commit on the 21st of Feb ([binaries here](https://github.com/jotego/jtbin/commit/7f1d60496bc7c3c0865506e3955815fed604266f) for jtcore's [...]

Using the rbf_r from the 21st Feb with today's JSON does not reproduce the problem

closed 2025-03-21 by jotego