jotego/jtcores #1013: outrun: Sound distorsion in Turbo Outrun

rp-jt, 2025-03-25:
> Hi, Here’s that short video clip so you can hear the sound distortion that happens at times after level 5 on Turbo Outrun. If it helps the corruption only ever happens once you have crashed your car. You then start to accelerate away again and you then get the distortion.

https://github.com/user-attachments/assets/633f6df7-e24b-428e-9c14-a028bc2e0636

Reported by RobH on discord

jotego, 2025-04-01:
@jtmiki check this and try to reproduce the problem at an earlier moment. Ideally in the game service menu. If not, try to reproduce this on level 1.

jtmiki, 2025-04-02:
I have checked it with today's compile all. Commit: 9313305
It is very difficult to describe.
From level 5, high-pitched instrumental sounds (cymbals?) are no longer audible and the low-pitched instrumental sounds (Floor tom or Rack tom?) are playing a sound that does not match the tempo.
I could not find this problem in levels 1-4.

jotego, 2025-04-04:
can you find that music in the service menu?

jotego, 2025-04-04:
related to #243

jtmiki, 2025-04-07:
I have heard all sounds of the service menu and No.3 PATCAR SIREN of EFECT B of SOUND TEST sounded strange.
Commit: 9313305