jotego/jtcores #1015: Mister Cores Too Wide in Direct Video

BrainGremlins, 2025-03-27:
So, I've noticed a strange issue when using direct video (240p over HDMI) on mister. On most cores it works perfectly fine, however, on some cores (specifically CPS1, 1.5 and 2 cores, that I have noticed, I'm not sure if others are affected), the output is an extremely wide 5600ish by 240 which seems to cause issues with the vast majority of DACs and switches that I have used. There was only one LCD monitor that it seemed to work with, which gave me the information on how wide the resolution is and even then, it was "out of range". Is this an issue with the cores themselves or is this something to do with the mister framework and if it is the former is there any way to fix it?

jotego, 2025-03-27:
The core uses the MiSTer framework as it is for this feature. Whatever improvements are done for this on the framework, we will port them here. 
There is nothing specific about this we can do.

closed 2025-03-27 by jotego

BrainGremlins, 2025-03-28:
Thank you for the response. I did ask about this on the mister github and was told it was specific to these cores, so not sure who to believe. That's okay, I'll just try to keep an eye out for equipment that this will work with, which just seems to be a bit rarer/more specialised 

Thanks again.

asturur, 2025-03-29:
For what is worth i m using those cores with DV with scalers without issues.
Probably the weird 384 horizontal resolution of cps1 and cps2 plays a role in this case, most of the cores have to multiply by 320 or 256

paulb-nl, 2025-03-30:
The horizontal output in Direct video mode is directly related to the video clock of the core. 

What can be done is reducing the video clock from 96MHz to 48MHz.