jotego/jtcores #1018: kchamp: right/left attacks mislabeled

rp-jt, 2025-03-31:
Reported on Discord by @WayNoWay

In Karate Champ, for Mister/Pocket, the default buttons are labeled as A="left (2)", B="right (2)", X="up (2)", Y="bottom (2)". However, pressing the button labeled as  "left" results in an attack to the right, and pressing "right" is an attack to the left.

Also testing it on Sidi shows a different arrangement for buttons: A="Left", B="up", X="right", Y="bottom"

jotego, 2025-03-31:
@jtmiki, please double check this first. If it fails, @rp-jt, please change the button order in the TOML in order to fix it.

jtmiki, 2025-04-01:
I have tested it and the left and right buttons work opposite to the default settings of MiSTer and Pocket.
The buttons work the same as the default setting in Sidi.
Commit: d1987a0

PR #1020 (2025-04-01): Swapped left-right button names for kchamp
Fixes #1018 by correcting the names of the assigned buttons
closed 2025-04-01 by jotego