jotego/jtcores #1019: flstory: support for Bronx

jotego, 2025-03-31:
The JTFLSTORY core supports games from MAME driver flstory.cpp and nycaptor.cpp. In the nycaptor driver, there are two systems: NY Captor and Cycle Shooting. NY Captor is working but Cycle Shooting is missing the ROM dump for the MCU. 

There is a working bootleg on MAME called Bronx. This one works partially on JTFLSTORY. There are some problems with the graphics or maybe with the system stability.

gyurco, 2025-04-03:
One problem is that the sub cpu is always in reset. Seems Bronx doesn't use reset control (Bronx' gametype is 1):
```
void nycaptor_state::cyclshtg_generic_control_w(uint8_t data)
{
.
.
  // shared palette data gets overwritten in colt without this
  if (m_gametype == 2)
   m_subcpu->set_input_line(INPUT_LINE_RESET, BIT(data, 1) ? CLEAR_LINE : ASSERT_LINE);
}

```

However allowing the sub-cpu to run (busrq control still in place) results in SRAM ERROR. Maybe shared RAM  handling has a problem? But it works in other games...

gyurco, 2025-04-03:
Actually it start sometimes (seems the sub-cpu overwrites some memory area during memory tests - needs to fix the enabling of the sub cpu), but then no improvement in graphics, even if the sub-cpu runs.

gyurco, 2025-04-03:
Looks like there's an 'user1' ROM section, which is mapped to the IORQ area in Bronx - commenting out in MAME has similar results in graphics what can be observed in the core!
Now my question is how to add two ROM regions to one CPU - one for MREQ, one for IORQ.

BTW, the sub-cpu halt (busrq) control doesn't make much sense - the RAM test writes arbitrary values into it, and if one value enables it, then it can interfere with the RAM test. The control resembles a watchdog write instead. But something must enable/disable the sub-cpu anyway...

gyurco, 2025-04-03:
I found out how to add the user1 region - hopefully I did it correctly. Now Bronx works.

jotego, 2025-04-04:
Thanks a lot!! @jtmiki this task is completed.

closed 2025-04-04 by jotego