jotego/jtcores #1022: cps and cps2 problems

leosmendes, 2025-04-01:
Hello good afternoon, I have been experiencing random crashes, game reboots and artifacts on the screen in CORE's of cps and cps2. I have not tested all the games for them, but in cps "street fighter 2 champion edition" and in cps 2 "street fighter alfa 2 and hiper street figter" the problems occur.
In champion edition, on the character selection screen the game crashed, the video and sound continued and it was still possible to hear the sound of inserting a coin, but nothing responded to the character selection confirmation. Also see artifacts in random places as per video 2 and 3. in cps2 "sf alfa 2" reboots, artifacts, error screen with debug. My memory works well at 141mhz, I use a lot of cores from n64, mega drive, sens, playstation without any apparent problems.

https://github.com/user-attachments/assets/89cf428d-3819-4412-aa65-1d4e15e91855
https://github.com/user-attachments/assets/17ee0ef3-4b75-4cb2-bf18-9b0fad42af91

jotego, 2025-04-02:
This is most likely a problem with your setup. Try using a different power supply, checkout whether you have power switches or thin power cables. Try removing, cleaning from dust and reinserting your memory module.

The memory test that you refer to is little significant as it uses the memory in a very specific way that is not representative of the way it is used on many cores.

I am going to leave JTCPS2 working for 12 hours today on one of our systems to double check this.

jotego, 2025-04-03:
Hello, I left SF2 Alpha running for 24 hours and it did not crash. As you already tried all the other recommendations, replacing the RAM module is the next step. When I made the JTCPS1 core this problem was rather common and it was only solved when new batches of RAM designed and built in a more robust way were available.

Fourtunately, RAM for the MiSTer has got much cheaper in the last year.

closed 2025-04-03 by jotego