jotego/jtcores #1031: CPS2 Xmen vs Street fighter, Marvel Super Heros vs Street Fighter and maybe other of this genre.

BilonFullHDemon, 2025-04-07:
When performing super attack, sprites are flickering with screen flickering effect, on success hit. This is not present on real hardware. On real hardware sprites are solid at all times. You can see it best when performing Wolverines super attack (QFC+LP+MP). NOTE: Maybe sprites are flickering on both instances but difference on FPGA vs Real Hardware is very very noticeable.

TheJesusFish, 2025-04-07:
I'm not seeing any sprite flicker on my HDMI setup.

What are you connected to? Also, if you have a video, that would be helpful.

BilonFullHDemon, 2025-04-08:
> I'm not seeing any sprite flicker on my HDMI setup.
> 
> What are you connected to? Also, if you have a video, that would be helpful.

Im on CRT. I made some videos and found out what's going on, it seams that on FPGA, background textures cover character sprites on some frames when super art hit oponent, which creates flickering effect.

[image]

BilonFullHDemon, 2025-04-08:
You can observe that better on slower super arts like for example F,QCF+punches with Wolverine rather than QFC+Punches that I motioned before which hits super fast.

BilonFullHDemon, 2025-04-08:
https://github.com/user-attachments/assets/ee1a8dbf-f82c-42c2-8c61-b72944a005fb

https://github.com/user-attachments/assets/3ce1fa01-4ed4-40c8-8b12-593da944aee2

BilonFullHDemon, 2025-04-08:
> is that visible during attract mode?

I will inspect that.

BilonFullHDemon, 2025-04-08:
One more comparison
FPGA

https://github.com/user-attachments/assets/0ed2d785-8e94-46b4-bc3d-88525ec4a000

REAL HARDWARE

https://github.com/user-attachments/assets/15f0f7ee-903e-4a3a-b730-4b52b5372599

BilonFullHDemon, 2025-04-08:
> is that visible during attract mode?

Yes.

TheJesusFish, 2025-04-08:
Still can't reproduce Wolverine, but Children of the Atom attract mode has other issues as well.

[image]

Double Ice man (courtesy of - on the mister discord)

BilonFullHDemon, 2025-04-08:
Another comparison but from LCD tv and in slow motion
FPGA
https://mega.nz/file/k71XkCTL#ZCEUK4P8Ci6MUiPRoomXbFiI4-66WbdhpKocxqGFYQk
This is the issue on FPGA if anyone don't notice (Background covering character sprites)
[image]

REAL HARDWARE
https://mega.nz/file/Quln0bRB#frwt61gSo4GP-FRCWqRxaK4wr8bH8V23VoMasgGlCqU

BilonFullHDemon, 2025-04-08:
Here is Xmen vs Street Fighter. I have no Real Hardware to compare with but Im 100% sure that this is not happening on original PCB
(Background covers Life Bar and characters sprites)
[image]

tdelage26, 2025-04-11:
I haven't got such a problem with my 2 MiSTer and also on MiST using CRT screens. I also own the original PCB and comparing it with FPGA gives the same result. Maybe the problem is linked with your screen 🤔

TheJesusFish, 2025-04-11:
Several people on the MiSTer discord have confirmed this bug.

BilonFullHDemon, 2025-04-11:
> Several people on the MiSTer discord have confirmed various rendering bugs.

Thank you for confirmation.

paulb-nl, 2025-05-10:
Here is a comparison of PCB and jtcps2.

X-men vs Street fighter 961023 USA - xmvsfu

It is the first fight of the attract intro.

PCB video: 
https://www.youtube.com/live/Y82kz69bphQ?t=6956

jtcps2 video:
https://github.com/user-attachments/assets/04399b4b-9fff-43e9-a63b-77e32d9f0217

PCB on top. Some priority issue with jtcps2. Mame has the same issue. It seems to happen always at the frame before a white screen. Maybe the game is trying to hide the sprites at the white screen and the priority is applied a frame too early?
[image]

closed 2025-12-30 by jotego