jotego/jtcores #1032: s16b: low music and sound effects

rp-jt, 2025-04-08:
> Yesterday I was playing Altered Beast on my Mister and the music and sound effects was very low. Same thing with E-Swat (System 16). Is that a know issue or a regular behavior? I played both games a few months ago and the sound volume was normal.
Tried Moonwalker (System 18) and the sound volume was normal too

- Reported on Discord by fabrisco

jtmiki, 2025-04-08:
I have tested following games in s16b with today's compilation, Commit: 43c3dd6. 

·Altered Beast
·E-Swat
·Alien Syndrome
·Bay Route
·Bullet
·Fantasy Zone II
·Wonder Boy III

Only E-Swat, the sound volume was a bit low. 

The 22/1/2025 compilation was also tested, Commit: 2ba30d5
Only E-Swat had a low volume in this compilation as well.

jotego, 2025-04-08:
change the volume for that machine in the TOML file. Run `jtframe mra -h` to see the details of how to do it.

rp-jt, 2025-04-09:
@jotego the volume for that machine is already set to 0xFF in the TOML file

jotego, 2025-04-13:
Please check that the volume settings are correctly read by the core.

jotego, 2025-04-21:
The volume information was present on the MRA, but the implementation of getting that to the sound mixer was missing. I completed it.

commit 124af87d (2025-04-21): support for game volume in MRA/JSON files. Fixes #1032
closed 2025-04-21 by jotego