jotego/jtcores #1037: NGP: Buttons do not work properly

jtmiki, 2025-04-14:
Buttons do not work properly in Melal Slug on NGP.
It moves forward on its own, cannot retreat, and cannot jump.
I have tried cores from the past and there may be a problem with cores from 30 August ,Commit: 324e62c or earlier.

jtmiki, 2025-04-15:
I have tested following games on NGP and NGPC.
·Metal Slug 
·Rockman
·Pac Man

Buttons do not work properly in Metal Slug and Rockman on NGP.
Cross-key and jump key do not work.

Contrary to this, buttons of these two games work well on NGPC.

Pac Man does not start on either NGP or NGPC.

I used today's compilation. Commit: 9902fe7

jotego, 2025-04-15:
Do you see this problem on other cores when using the Analogue Pocket?

aminta, 2025-04-18:
I noticed a problem in Ngpc Neo Turf Masters, only left arrow works when choosing shot effect before shooting. In both Ngp and Ngpc you can’t  change the controls in menu,

closed 2025-07-09 by jotego

rp-jt, 2025-07-09:
The problem with the controls should be fixed in 09e08c610d01334021bda96d6365fc6d2f7b74a4

I will include Pac-Man not booting up in #788