jotego/jtcores #1040: Support for Controller Alignment

rp-jt, 2025-04-22:
For multiplayer games, it could be possible to implement a way to assign in which player slot each controller goes.

This would be particularly useful for games where the character you use depends on what Player number you are playing as.

It could be added to the framework and being enabled by a macro.

jotego, 2025-06-08:
Implemented in e29b452

jotego, 2025-06-08:
@jtmiki please test character selection via OSD for four-player games on JTTMNT,JTRIDERS and JTSIMSON.

closed 2025-06-08 by jotego

jtmiki, 2025-06-09:
I have tested them and I could select four different characters and could play well.
If you select another character while playing alone and insert a coin, two characters will appear on the screen.
In TMNT you can only control one character, but in SUNSET RIDERS and SIMPSONS the movements of two characters are synchronized.

[image]

[image]

Commit: e29b452