jotego/jtcores #1048: Second Level - Misalignement of sprites with background

funkycochise, 2025-04-26:
I have several pictures.
I encounter the problem in the stairs
at some point the hero is walking in the background

[url=https://gamoovernet.pixhotel.fr/pics/v2/2025/04/26/20250426111113-funkycochise-IMG_5591.png][img]https://gamoovernet.pixhotel.fr/pics_gamoovernet890px/v2/2025/04/26/20250426111113-funkycochise-IMG_5591.png[/img][/url]
[url=https://gamoovernet.pixhotel.fr/pics/v2/2025/04/26/20250426111158-funkycochise-IMG_5592.png][img]https://gamoovernet.pixhotel.fr/pics_gamoovernet890px/v2/2025/04/26/20250426111158-funkycochise-IMG_5592.png[/img][/url]
[url=https://gamoovernet.pixhotel.fr/pics/v2/2025/04/26/20250426111215-funkycochise-IMG_5593.png][img]https://gamoovernet.pixhotel.fr/pics_gamoovernet890px/v2/2025/04/26/20250426111215-funkycochise-IMG_5593.png[/img][/url]

f3bandit, 2025-04-27:
same issue as well on mister and analogue pocket

f3bandit, 2025-04-27:
[image]
[image]
[image]
[image]

jotego, 2025-05-04:
Might be related to #1042

jotego, 2025-05-17:
Fixed in a30a278cd

closed 2025-05-17 by jotego