jotego/jtcores #1049: Diagonal input problem on Rolling Thunder (rev 3)

runeapetersen, 2025-04-27:
This is in relation to the the Rolling Thunder MiSTer beta. Tested on Rolling Thunder (Rev 3).

When crouching while facing right, the character will not turn around when switching input to diagonal left/down. Crouching while facing left and inputting down/right diagonal will turn and walk the character to the right as expected. The latter is similar to how MAME works when moving the character in the opposite direction from a crouched position (in both directions).

funkycochise, 2025-04-28:
For me it's not possible to turn around when crouched. The character will stand up.
I need to verify again on the PCB though

runeapetersen, 2025-04-28:
To elaborate what I'm seeing in the MiSTer core:
Case 1: 
1.1 Character is crouched, facing left. Input is ⬇.
1.2 User changes input to ↘
Result: Character starts walking to the right. This is as expected from playing the core in MAME.

Case 2:
2.1 Character is crouched, facing right. Input is ⬇.
2.2 User changes input to ↙️.
Result: **Nothing happens. This is not as expected.** In MAME the character starts walking to the left.

Since Rolling Thunder is apparently a 4-way game without diagonals, I guess Case 2 is the correct behaviour for both directions (how the PCB would work). Right now it's a mix of PCB and MAME, which seems like an error.

jotego, 2025-04-28:
@jtmiki please check this on the original PCB

jtmiki, 2025-04-30:
I have tested it on the original PCB and the character does not move when he is crouched facing left or facing right. 
Commit: 7ba88ac

https://github.com/user-attachments/assets/d446db48-a509-487d-86f5-c49b0210de57