jotego/jtcores #1053: [jtcircus] Jumping for a 4th time in a drum in 3rd level doesn't show the clown trespassing the circus tent

c0de90e7, 2025-05-09:
English:

On the third stage ( and maybe other stages ), the clown is jumping from drum to drum to advance. If a jump on any given drum is repeated in the same drum, the clown gains momentum and jumps a bit higher each time. This is useful to catch the money bags that appear above the top of the elephant doors height. You jump three times over the same drum so you make the clown reach those money bags. However, if you jump for a fourth time, the jump is too high and the clown will trespass the circus tent at the top ( the yellow and blue one ) and a sprite representing the clown's face breaking the tent plastic will appear. This doesn't happen in the jtcircus core ( seen in Analogue Pocket ).

https://youtu.be/Z2QPrSMYPx8

Please watch at 03:55, there's four jumps from that point:
- the first one, arriving from the previous drum,
- then a couple more jumps which make it jump higher but are still fine
- and a fourth jump that makes it jump too far/high and the face trespassing the broken tent plastic sprite appears 

Spanish:

En el tercer nivel ( y quiza en otros niveles ), el payaso salta de tambor a tambor para avanzar. Si un salto en cualquier tambor se repite en ese mismo tambor, el payaso gana impulso y salta algo mas cada vez. Esto es util para alcanzar las bolsas de dinero que aparecen por encima de la altura de las puertas de los elephantes. Saltas tres veces encima del mismo tambor asi que haces que el payaso alcance esas bolsas de dinero. Sin embargo, si saltas una cuarta vez, el salto es demasiado alto y el payaso traspasa la tienda del circo que esta arriba del todo ( la de color amarillo y azul ) y aparecera el sprite que representa la cara del payaso rompiendo el plastico de la tienda. Esto no ocurre en el core jtcircus ( visto en la Analogue Pocket ).

https://youtu.be/Z2QPrSMYPx8

Por favor, vean en 03:55, hay cuatro saltos desde ese punto:
- el primero, llegando desde el tambor anterior,
- luego un par de saltos mas que lo hacen saltar mas alto pero todavia estan bien
- y un cuarto salto que lo hace saltar demasiado lejos/alto y aparece el sprite de la cara traspasando el plastico roto de la tienda

jotego, 2025-06-08:
Fixed in [circusc branch](https://github.com/jotego/jtcores/tree/circusc).
@rp-jt, in order to make this fix I had to add scene support to JTCIRCUS. That required taking the RAM out of the jtkicker_scroll module. That module is used on the cores:

- [ ] kicker
- [ ] mikie
- [ ] roc
- [ ] sbaskt
- [ ] yiear

Please change these cores by adding the scroll VRAM to mem.yaml. You do not need to add full scene support to them. Just fixing them after the change is enough for now.

jtmiki, 2025-06-10:
I have tested a game of each core, there are six cores in total.
All of them worked well.