jotego/jtcores #1054: [jtsimson] Controls option is not accessible for Analogue Pocket

go2sleep327, 2025-05-10:
On the latest update (5/10/25) for the "The Simpsons" arcade game core on the OpenFPGA for the Analogue Pocket, the player can not configure the controls/

1. Press Analogue button to enter the Home Menu
2. Go to Core Settings
3. Go to Controls
4. Attempt to select Controls

Also, hi this is my first time reporting an issue here, so I apologize if I'm doing it wrong, which I'm sure I am, any advice on how to report a bug would be helpful. Love Jotego's work!!

jotego, 2025-05-11:
@jtmiki please check this

jtmiki, 2025-05-12:
This is the confirm button on the analogue pocket.

[image]

To change the button setting, first press the confirm button in Attack/Jump and then select the Attack/Jump button you want to use.

I have tested it on today's compilation (Commit: ae4b249), there were no problems with the button settings.

go2sleep327, 2025-05-12:
I'm sorry, i forgot to mention i was using the "The Simpsons (2 Players World, set 1)". The button settings work on "The Simpson (4 Players World, set 1) as well as the alternatives. I apologize for the inconvenience if this is not an issue.

rp-jt, 2025-05-14:
Tested on ae4b2498e10140f6fa0e202faaf9669ad5519d4a for "The Simpsons (2 Players World, set 1)" and the remapping controls settings works correctly

@go2sleep327 if the problem is persistent, try removing the simson files from your sd card and upload them again from the latest update, in case there is some file that shouldn't be there anymore creating interference

closed 2025-05-14 by rp-jt