jotego/jtcores #1058: s18/lghost: fix missing sprites

rp-jt, 2025-05-19:


jotego, 2025-05-19:
Should there be more bits for obj_bank?

rp-jt, 2025-05-19:
> Should there be more bits for obj_bank?

`obj_bank` is coming from `tbl_dout[11:8]` in this case (assigned in `jts16_obj_scan`). 
In simulation, it looks like `tbl_dout[13:12]` are empty when `obj_bank` is registered

Also, at first I tried modifying only the higher bits of the address and it wouldn't work, I had to modify the bank section as well to get the image right. The original value obtained from `obj_bank` in this scene was `obj_addr[22:17] = 6'h0B`,  in contrast with `obj_addr[22:17] = 6'h33` (which shows the correct sprite)

jotego, 2025-05-19:
Look at the MRA file. Maybe the ROM order can be changed so things fall in the right place.

PR #1060 (2025-05-20): s18/lghost: sprites correctly addressed
Fixes #1058
closed 2025-05-20 by jotego