jotego/jtcores #1063: Rolling thunder & pacland inputs

shuffle5, 2025-05-24:
When I assign the buttons for the last 3 namco cores including Rolling T and now pacland I cannot get the core to register the buttons . I have the latest jtbeta key that came with pacland . I cannot start or play once they have loaded up

jtmiki, 2025-05-27:
I have tested following games with changing the button settings on MiSTer and Pocket.
They worked well.

- Rolling Thunder
- Wonder Momo
- Sky Kid Deluxe
- Pac-Land

I used today's compilation. Commit: f5a0df6

shuffle5, 2025-05-27:
I'm using a 8bitdo m30 controller and also tried using a Xbox one too. Can confirm all those namco  games are unable to accept inputs still. If I press one of the buttons it will pause that game . So at a loss to why . Have tested other games all ok

shuffle5, 2025-05-30:
Yes I will ask in discord as know you are busy . But indeed you are right when I load up sega system 18 , & sunset riders I get the same thing . These load up with the credits rolling down the screen and a yellow background but I am a Patreon subscriber so not sure why...  and I have the pacland jtbeta key from 23/05/25 in the games/mane folder.

jotego, 2025-06-01:
Maybe you have unzipped the jtbeta.zip file? It should be there unzipped. There is a short FAQ [here](https://github.com/jotego/jtbin/wiki/Beta-Files-FAQ)

closed 2025-06-01 by jotego