jotego/jtcores #1098: CPS cores loading very slow in last release

rp-jt, 2025-06-25:
Several users report that CPS1, CPS1.5 and CPS2 cores are loading slower than they used to since latest release  (https://github.com/jotego/jtbin/commit/3ceea61c84c0f21db9d33208889b8500dfb45f7e)

rp-jt, 2025-06-25:
@jtmiki Could you compare the latest release with an older one to see the difference? Thanks!

jtmiki, 2025-06-25:
I have tested following games of CPS1, CPS15 and CPS2 with last Friday's release.
The results are so clear that there is no need to compare them with others.

**CPS1**
1941
Forgotten Worlds
Mercs
SF Zero

**CPS15**
Cadillacs and Dinosaurs
Saturday Night Slam Masters
The Punisher

**CPS2**
19XX
Dimahoo
Eco Fighters
Marvel Super Heroes
SF Alpha 2
X-Men Children of the Atom

The loading time of CPS1 games are acceptable.
CPS15 games take a little while to load.
But CPS2 games take a ridiculously long to load.

Toryalai1, 2025-06-25:
Do we know if fast ddr loading is being used?

tdelage26, 2025-06-26:
> New MRA:
> 
> rom index="0" zip="ssf2.zip|qsound.zip" md5="None" asm_md5="ecf1e3a355748994730378115c4b33b2"
> 
> Old MRA:
> 
> rom index="0" zip="ssf2.zip|qsound.zip" type="merged" md5="None" address="0x30000000"
> 
> it is missing fast ddr loading in the mra's

Yes, that's just an RMA problem, the core is okay... replacing the MRAs from last friday update with the ones from the week before fix the problem ;)

commit b5002d2b (2025-06-27): DDR load restored for CPSx cores. Fixes #1098
closed 2025-06-27 by jotego