jotego/jtcores #1104: Pac-Land - Round 19/30 GFX effect not displaying properly

ExMosquito, 2025-07-05:
This core is pretty much perfect aside from a game breaking graphical effect on Rounds 19 and 30. On the PCB/Mame these rounds have a headlamp effect that uncovers a section of the screen depending on what direction you’re facing. At the moment this isn’t implemented, making the stages unplayable and almost impossible to navigate. I really hope this can be fixed because otherwise the core is great. But at the moments if you’re half-decent at PacLand you cant play a full credit. 

https://ibb.co/BH1Y4k03

Thanks!

jotego, 2025-07-06:
Thank for reporting this. It is a bit hard to get to that level for us. Could you please do this?

- play on MiSTer up to a point where the GFX are broken on round 19
- then go to the OSD menu, move to the system part of it and select **Save Settings**
- copy the file `Pac-Land (World).nvm` your MiSTer SD card. It is in folder `config/nvram`
- attach that file to this issue

We can then simulate exactly that scene on our setup and debug it very quickly.

Thank you!

jotego, 2025-07-07:
A small change that made the game playable was made in af348de. The colmix needs a major redesign to fit the original PCB more closely. Work in progress in branch [pacland-colmix](https://github.com/jotego/jtcores/tree/pacland-colmix).

closed 2026-09-27 by jotego