jotego/jtcores #1122: Broken link to jtframe in readme

smdjeff, 2025-08-03:
Several core readme's say  "The core is compiled using jtcore from JTFRAME... https://github.com/jotego/jtframe" but that is a 404.

closed 2025-08-04 by jotego

smdjeff, 2025-08-04:
> Thanks. It's fixed now.

Thank you.  Btw, I can't find the sys_top.v I'm interested in seeing how
you're routing video so I can adjust it and rebuild it for my board.
https://github.com/smdjeff/DE10Jamma
I'm also working on an even more hardcore version for CRY vector displays
like Tempest and Sega G80 Star Trek.

jotego, 2025-08-05:
Please read the abundant doc files in the project to understand the build process. 
You should create a new JTFRAME target in the right location for your system. Look at the existing ones to learn what's needed and how things are reused from one target to another similar one.

commit 2780f918 (2025-08-04): Fixes broken link. Fix #1122