jotego/jtcores #1127: Wonder Planet graphical issues

wwark, 2025-08-07:
See the screenshot !

[image]

Thanks

jotego, 2025-08-11:
Please @wwark check that you are using the latest combination of .rom .mra and .rbf files

wwark, 2025-08-11:
I found that is the issue after removing the dip . the behavior is correct.
If I set flip screen you will have this screen.
 
[image]

Could you please reopen the issue ?

jtmiki, 2025-08-12:
As wwark said if dip flip screen sets ON, there are graphical problems.

[image]