jotego/jtcores #1163: Consider how to make analogizer part of Pocket target public

jotego, 2025-09-01:
TotalFury has been improving the analogizer and he would like to contribute his changes, but our files are part of the Pocket target, which is held private. 

Consider how to make the analogizer section of the Pocket target public so we can take PR's.

sduponch, 2025-09-01:
Thank a lot for considering my request :)

The Analogue Pocket framework does not provide "Arcades" system related inputs like Test, Service and Coins and my JVS module will be able to provide theses signals. So we will need map theses with the Analogizer_SNAC module.

jotego, 2025-09-20:
Thanks for sharing data. We didn't have the time to look into this last week. We'll update on this topic soon.

jotego, 2025-09-22:
@sduponch my understanding is that you want to be able to interface the Pocket with a JVS system. I only know that the JVS system is a serial protocol to transfer video, (audio?) and player input signals. Is that so?

Can you do that without breaking the core for regular users and also for users of the Analogizer in the original way? Or do you require two different builds?

Have you checked how we implemented analogizer support from the user's point of view? The configuration done on the PC first, etc?

sduponch, 2025-09-24:
Here a basic example on how i can integrate JVS with the Analogizer NeoGeo core : 

https://github.com/sduponch/Analogizer_Mazamars312_Neogeo_0.8.1/commit/1e3f786d27549c936cba8ccaf28932b546aff351

closed 2025-09-25 by jotego

sduponch, 2025-09-25:
Ok i made a false start !

I have two questions : 
- Where is located the yaml file to add dependant files ? i have greped the whole repository after "analogizer" and did not any yaml file. maybe it is located on the private pocket folder.
- What is the procedure to trigger the build action ? It looks like it is branch related, but i don't think i have rights to create a branch on this repo.

rp-jt, 2025-09-26:
> I have just a little question, how are mapped the coins outputs ? Are-they accessible via "snac_p[1-4]" or they are not mapped yet ?

The coin and start buttons are assigned as bits 14 and 15, respectively, at snac_[1-4]
There is an automatic assignment for the SNAC controllers if mixed with the USB ones as explained [here](https://github.com/jotego/jtbin/wiki/Analogue-Pocket-Cores#automatic-controller-assignment)

After this assignment, the controller signals are converted into the JTFRAME format and then sent to [jtframe_board.v](https://github.com/jotego/jtcores/blob/master/modules/jtframe/hdl/jtframe_board.v)
In this module, we could say `snac_p[1-4] -> board_joy[1-4]` 

Also, I am aware that, right now, we do not have output ports at `jtframe_analogizer.v` to let out the controllers analog joystick signals into the framework. Let us know when you have a working version without them and we can add them to the Pocket submodule

sduponch, 2025-09-27:
So the remaining issues are :  
- snac_cont looks truncated (maybe i did not map snac.options correctly)
- missing coins outputs
- missing analog outputs (quid of drivings games like Outrun, how to cleanly map steering, gas and brake ?)
   maybe we could use #define to specify the max numbers of players, button per player, analog mapping
- missing test/service outputs
- my fat JVS module does not fit on the s18 core :\

rp-jt, 2025-09-30:
A couple of comments about the issues:

- `snac_cont` IS truncated because the input port in the Analogizer_SNAC module has only 5 bits. `jtframe_analogizer.v` receives 8 bits from the setup file, so you just need to pass more bits to the next module. Also, I see that you have a specific number in that port in your branch, so the value in the setup file can't be read by the SNAC module
- I find it odd about the coins and inputs in CPS. Have you checked if this also happens in the **Release** version of the cores? In that case, we should open an issue about it
- There are missing options in the setup python script because we wanted to make it simpler in the jtcores adaptation of Analogizer. I would recommend you build around the existing options and configuration if possible, so it's easier to understand and more cohesive with the rest of the repository for the case of a PR. 
About this, I also see that you replaced our implementation of `openFPGA_Pocket_Analogizer_SNAC.v`. This one was simplified in order to fit our format, pass our tests and linter... but it has all the information of the original at the point of merging. While evaluating the changes made, it is easier to check small changes to this document (that should be working already within our framework) adding the new features rather than checking a whole new file in the repository. This might also help to pinpoint where some issues might be coming from. I know this option takes some time, but I think it can be worth it!
- The analog output not being rotated in vertical games like the digital pocket screen or the hdmi output is not an issue. That's how it works as well with the other FPGA platforms, since it matches the original operation of Arcade machines, where the whole screen was rotated for these games

I hope this helps! Let us know if you have any other doubts

sduponch, 2025-09-30:
Thank you for your feedback @rp-jt !

- To test my JVS module i have "hardcoded" the snac_cont value but this is not the case in the "analogizer_update" branch. I will split changes to the "openFPGA_Pocket_Analogizer_SNAC.(s)v" in more commits to be cleaner in what has been changed/updated/added.
- Unfortunatly i do not have other SNAC interface than my custom JVS adapter, RndMnkIII the creator of the Analogizer will ship me soon a SNAC module to confirm the issue with CPS[1-2] and to tests the PSX code update.
- I asked RndMnkIII to add a screen rotate option to the Analogizer, no more torticoli when playing contra.

Can you take a look at routing joystick (analog) and coin inputs ? in fact, i can add credits using Pocket button until my interface start inputs polling, after that, pocket (credit) buttons became unresponsive.

sduponch, 2025-10-01:
Thank a lot for you reply, i will dig more about this issue.

So the pocket buttons are active until the core receive something from the SNAC, that why it became inactive just after my JVS start polling. I think this is maybe just a buttons mapping issue, i don't know on what btn bit the coin/start buttons are mapped. What is strange is that it works for all cores except for the CPS ones (and they are my favorite obviously).

jotego, 2025-10-03:
@sduponch I need to cancel your action runs today because I need GitHub to run our release actions for making the weekly update to patrons. Sorry about that. This may happen on Fridays as it is when we make new releases.

sduponch, 2025-10-03:
Ho no problem !

robibennobi, 2026-04-20:
This may not be the right place for this question, was just trying out snac with the psx namco stick on jotego cps2 pocket core this evening see if it works and didn't see any option in the pupdate for psx controllers in his anlogizer settings, is it just the standard anolgizer config supports psx at the moment not jotego cores