jotego/jtcores #1209: Wonderboy III Black Sprite and background colour issue

niknakniknak, 2025-09-30:
This is happens usually after running a bunch of other cores, but you can also sort of occasionally force it by launching the wb3 core from a cold boot and then within wb3 launch the core again a couple of times whilst wiggling the joystick as it loads the core.

When credited up you get an incorrect blue background

[image]

Then when getting to an end boss it is black

[image]

real-amano, 2025-09-30:
Are those effects always connecte? Thus if you see the blue screen you know  that the boss will be black?

niknakniknak, 2025-09-30:
> Are those effects always connecte? Thus if you see the blue screen you know that the boss will be black?

It seems that way, I've never noticed black sprites without the blue.. but I can't be 100% on that.